prusa3d/PrusaSlicer #6935: start gcode M907 can damage printer if print settings not named as specified?

Rayman44, 2021-09-12:
### Version
Version 2.3.0+win64

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
MK3s

### Behavior
if my saved print settings is not named DETAIL @MK3 or QUALITY @MK3 or SPEED @MK3 or DRAFT @MK3
then it will damge my printer. Yes?

start gcode:
; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}

also what is the default extruder current?
and what if i use M907 E430 all the time?

#### Project File (.3MF) where problem occurs
NA

rtyr, 2021-09-12:
> if my saved print settings is not named DETAIL @mk3 or QUALITY @mk3 or SPEED @mk3 or DRAFT @mk3
> then it will damge my printer. Yes?

No. 

> also what is the default extruder current?

538

Rayman44, 2021-09-12:
OK. I belive the gcode lines are to prevent damage.
But only does so if DETAIL @MK3 or QUALITY @MK3 is used. If the print setings is the same and saved with diferent name. Then defalut current is used = E values changed from 430 to 538.

rtyr, 2021-09-12:
Those lines are there to reduce extruder motor current for slower system profiles. The goal is to reduce extruder motor temperature when high current is not needed.

The warning is about E values. (to not use different E values).

I believe we can close this.

Rayman44, 2021-09-12:
so the printer is only safe on 430 or 538 and nothing else? say like 450?
Then this M907 setting is better as a tickbox in the print settings tab.

rtyr, 2021-09-12:
These values were tested/chosen. 450 should be safe for the printer. It is not usual to use such command in start gcode, we may implement automatic current control into the firmware. It is certainly not something which should be in the settings.

Closing.

closed 2021-09-12 by rtyr