prusa3d/PrusaSlicer #6936: Refurbation / Correction Bridges

gogsi1, 2021-09-12:
### Version
2.4.0-alpha1
P.s. Version is not important. This is in all versions. 

### 3D printer brand / version + firmware version (if known)
It does not play a big role. Problem in the implementation of the Slicer function 

### Behavior
When printing with a smaller filling 100%, always the first top layer of the cover is defined as a bridge.
It is necessary to fix it or make it determined by the example of as an "additional bridge" and to be able to adjust the speed and blowing.

rtyr, 2021-09-12:
Why? Please provide more detailed description.

gogsi1, 2021-09-12:
Why does he do it? 
[image]
In such places, there is enough solid infill.

rtyr, 2021-09-12:
Why is it using bridge infill when it bridges over sparse infill? Why is that problem? In alpha1 you can switch bridge infill to use regular flow ("thick bridges" parameter).

gogsi1, 2021-09-13:
This is what if filling less than 100%. 90% This is as an example. This layer should not reduce the speed of printing and reduce speed.

gogsi1, 2021-09-13:
Maybe I will explain crookedly, I want to say. In this place, it should not reduce speed.
Should not turn on cooling