prusa3d/PrusaSlicer #6939: Fatal error on 2.4.0a1 Win32

Tobor7, 2021-09-13:
### Version
2.4.0 alpha1+win32

### Operating system type + version
Windows 10 Pro version 21H1
Ram 16 Go

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
After the slicing the file GOTHIC_LITO_BASE.3mf, I got this error (Please note that I don't get this error with the win64 version) :
[image]

Memory 5 seconds before the error:

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[GOTHIC_LITO_BASE.zip](https://github.com/prusa3d/PrusaSlicer/files/7154407/GOTHIC_LITO_BASE.zip)

bubnikv, 2021-09-13:
Please run a 64bit build of PrusaSlicer if possible. You are running a
32bit build.

Tobor7, 2021-09-20:
The project is working fine in the 2.4.0a1-win64 version but was also working fine in the 32bits 2.3.3. Only got this error with the 2.4.0a1 win32