prusa3d/PrusaSlicer #6940: time estimates are missing on gcode files created for the MINI (2.4.0-alpha1)

qorron, 2021-09-13:
### Version
2.4.0-alpha1+linux-x64-202109011425.AppImage

### Operating system type + version
Debian Testing

### 3D printer brand / version + firmware version (if known)
Prusa MINI (upgraded to +) 4.3.2

### Behavior
checked out PS 2.4 alpha because it has improvements I could use right now.
noticed that the config was empty because it uses a different location to store it so: wizard, add my printer and some filament profiles.
open an older 3mf file with parts in it, notice that some advanced settings are no linger the defaults, so start a new project, reset everything to defaults, add the stl files (thus avoiding a 2.3.3 3mf file), add support, brim to one of the parts, slice, plug in usb, store gcode on usb, unmount, put usb in printer -> observe screenshot.
Print Time: unknown
Material: PLA/0 g/0.0 m

gcode file made with 2.3.3 still show time and material estimates.

#### Project File (.3MF) where problem occurs
[Beer_Cap_Gun_left_half_32mm_v2_2.4.zip](https://github.com/prusa3d/PrusaSlicer/files/7156060/Beer_Cap_Gun_left_half_32mm_v2_2.4.zip)
[image]

#### Note
I think the MINI prints faster than usual with 2.4a than it did with 2.3.3
had some warping and bed adhesion issues but those were all fixed by rejuvenating the smooth sheet with acetone. imho both observations point to an issue in the MINI profile in 2.4a

I hope this is no duplicate (I searched but found no matching report) and you are able to reproduce the issue.
thank you,
-q

rtyr, 2021-09-13:
It seems to be caused by the fact that the firmware looks for these values ​​in the last 10000 characters of the g-code file. Comments in g-code file generated by 2.4.0-alpha1 have more than 10k characters. We will discuss how to fix it (PrusaSlicer or firmware). Thank you for reporting it.

> I think the MINI prints faster than usual with 2.4a than it did with 2.3.3

Print speeds are the same, but travel acceleration is significantly higher (twice as high currently). It is indended, 2.4.0-alpha supports separate setting for travel acceleration, unlike older PS versions. It indeed "feels" faster overall.

rtyr, 2021-09-15:
It will be fixed in the firmware. Closing.

closed 2021-09-15 by rtyr