prusa3d/PrusaSlicer #6944: Color change symbol not shown in Gcode Preview

laptop-laptop, 2021-09-15:
### Version
2.3.3

_Use `About->About PrusaSlicer` for release versions_
2.3.3

### Operating system type + version
_What OS are you using, and state any version

Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior

Take any model and pick any layer. Then, issue a Color Change using the + sign on the vertical slider.
Export the gcode and then load it via File->Gcode Preview. Then, go to that layer and the Color Change circular symbol is not shown. Is this normal? How can I make the Preview to show the Color Change symbol to verify that if I send it to the printer, it will pause at that layer for filament change?

rtyr, 2021-09-15:
It is not visualized properly in 2.3.3, but you can see the color change if you switch View to "Color print". It is already fixed in 2.4.0-alpha 1.

2.3.3
[image]

2.4.0-alpha1
[image]

laptop-laptop, 2021-09-15:
Is there any way to show the exact position where the Color Change M600 command is executed?

In View: Color Print and Show: Color changes, a donut-like symbol is supposed to show that but sometimes it is not shown or in the wrong location.

rtyr, 2021-09-15:
> In View: Color Print and Show: Color changes, a donut-like symbol is supposed to show that but sometimes it is not shown or in the wrong location.

Please provide some 3MF project files, where this problem occurs.