prusa3d/PrusaSlicer #6945: Height Range Modifier changes not showing up in gcode

laptop-laptop, 2021-09-15:
### Version
2.3.3

_Use `About->About PrusaSlicer` for release versions_
2.3.3

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

i3MK3S+

### Behavior
I used the Height Range Modifier to make some modifications of settings in some layers. However, they don't show up in the gcode. Where does PrusaSlicer store those extra modifications made via Height Range Modifier? Is it in a special section within the gcode? Is there a delimiter that I can use to search for those modifications I made?

lukasmatena, 2021-09-15:
Please post your 3MF.

> Where does PrusaSlicer store those extra modifications made via Height Range Modifier?

You mean in the gcode? Nowhere. It just uses the settings.

laptop-laptop, 2021-09-15:
This is just a general question, not related to any specific 3MF.

If such additional settings made under the Height Range Modifier are not stored in the gcode, how the printer knows about these changes given that the printer just follows the commands from the gcode.

lukasmatena, 2021-09-15:
Because slicer uses the settings on those layers and generates different commands. If you set e.g. perimeters to 2 by default and change it to 3 in a layer range, slicer will generate different gcode - one with three perimeters on those layers.

I understood that you were asking whether there is some comment header in the gcode saying "from now on, I'm printing with 3 perimeters instead of 2". The answer is no.

laptop-laptop, 2021-09-15:
Thank you for the clarification. So the slider takes those modifier changes into consideration when generating the gcode without having a separate section listing explicitly the specific changes. I was hoping that I could just turn on/off some modifiers or change the values by quickly editing the gcode without slicing and re-editing the gcode again.

Speaking of comment header, usually in between ;BEFORE_LAYER_CHANGE and ;AFTER_LAYER_CHANGE, there is a block of codes between ;WIPE_START and WIPE_END. What is Wipe? What do those codes in between do?

lukasmatena, 2021-09-15:
Install [PrusaSlicer 2.4.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1). When you slice and start moving the horizontal slider in preview, it will show a gcode snippet and move the tool accordingly. Wipe is an extra move before travel, its purpose is to reduce stringing. It can be turned on/off in Printer Settings->Extruder->Wipe while retracting.

laptop-laptop, 2021-09-16:
Thank you for telling me such nice feature. From the gcode, it looks like during Wipe, the extruder also moves when a negative value is given to E to retract. Is it to save time? Under what conditions does the slicer create the WIPE_START and WIPE_END?

closed 2026-09-29 by prusaslicer-bot[bot]