prusa3d/PrusaSlicer #6946: M601 not removed for new model

ceesse, 2021-09-15:
### Version
Version 2.3.0+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S + MMU2S

### Behavior
After loading a new model, PAUSE M601 is still enabled from previous model. This is really annoying. I suggest to clean those pauses when loading a new model.

rtyr, 2021-09-15:
> I suggest to clean those pauses when loading a new model.

Which will be really annoying for somebody else (and it doesn't make sense unless all objects are deleted).

Use "Delete all" or "New project", it will do what do you need.

fe60, 2021-09-15:
If you remember the leftover from previous model(s) after importing new one(s), there is a "Discard...changes" button at the bottom of the vertical slider of the print preview.

[image]

rtyr, 2021-09-16:
Answered. Closing.

closed 2021-09-16 by rtyr