prusa3d/PrusaSlicer #6949: "The print is empty" when trying to print small object

meermanr, 2021-09-15:
### Version

PrusaSlicer: Version: 2.4.0-alpha1+x64
Build: PrusaSlicer-2.4.0-alpha1+x64-202109011420
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOs Big Sur Version 11.4 (Build 20F71)
Total RAM size [MB]: 68,719MB

### Operating system type + version
MacOS 11.4.0

### 3D printer brand / version + firmware version (if known)
Prusa MINI+ (stock)
Firmware 4.3.2 
Bootloader 1.1.0
Buddy board 1.0.5 

### Behavior

[image]

Small (dimensions) and simple (low triangle count) STL file result in error:

> PrusaSlicer has encountered an error
> Flow :spacing() produced negative spacing. Did you set some extrusion width too small?

a notification in the bottom-right of the 3D editor view appears which reads:

> WARNING:
> Empty layer detected between heights 0.2 and 6. Make
> sure the object is printable.
> Object name: Alignment tools Dowel.stl
> This is usually caused by negligibly small extrusions or
> by a faulty model. Try to repair the model or change its
> orientation on the bed.

Changing the orientation results in a different error notification in the bottom right of the 3D Editor View:

> ERROR:
> The print is empty. The model is not printable with
> current print settings.

I've tried changing various printer settings (nozzle sizes) and print settings (layer heights, number of top and bottom shells, etc) , and the "best" result I can get is a skirt being printed, and **only** the top layer - no layers between the build plate and the top layer!

Steps to reproduce:

1. Launch Prusa Slicer, which presents with the following system default settings:
> Printer: Original Prusa MINI & MINI+ (stock profile)
> Print settings: 0.20mm QUALITY (stock profile)
> Filament: Prusament PLA (stock profile)
2. Drag'n'drop the attached "Alignment tool - Dowel.stl" file to the build plate
3. Observe errors, as described above

**Is this a new feature request?**: No.

#### Project File (.3MF) where problem occurs

[3MF, STL, and screenshots.zip](https://github.com/prusa3d/PrusaSlicer/files/7171193/3MF.STL.and.screenshots.zip)

rtyr, 2021-09-15:
Probably some issue with the model. I was able to reproduce it on master, it was sliced OK after Netfabb repair though.

Cura and Simplify3D also failed to slice provided model.

Repaired model attached.
[Alignment tools - Dowel_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/7172941/Alignment.tools.-.Dowel_fixed.zip)

meermanr, 2021-09-15:
Thank you - by comparing the two in Blender I found the issue: there is a duplicate mesh, like russian dolls, one slightly smaller inside the other.

🤦

closed 2021-09-15 by meermanr