prusa3d/PrusaSlicer #6953: Printer Retraction placeholders are using the last known Filament Override values

cloud30000, 2021-09-15:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
N/A (Tested with Prusa MK3S+ profile, Anycubic Mega Zero profile, and Anycubic Vyper (custom Mega) profile

### Behavior
The "retraction_XXXXX" scaler placeholders as well as the "retraction_XXXXX[0]" vector placeholders are using the last known Filament Override values, both when the Filament Override is selected and when they are disabled ("filament_retraction_length" is nil, but "retraction_length" is still the last known numeric value).

These placeholders should instead be using the Printer profile values.

**Printer** Settings Extruder 1:  **Retraction Length 7mm, Lift Z 0.3mm**
**Filament** Settings Filament Overrides ("-default-" selected, all boxes unchecked): **Length N/A, Lift Z N/A, Retraction Speed N/A**
Start G-code: "`G1 E-[retract_length] Z-[retract_lift] F{retract_speed[0] * 60}`"
results in the following inside G-code file:
**_G1 E-1 Z-0.2 F3900_**

**Printer** Settings Extruder 1:  **Retraction Length 7mm, Lift Z 0.3mm**
**Filament** Settings Filament Overrides ("-default-" all boxes checked): **Length 2mm, Lift Z 0mm, Retraction Speed 40mm/s**
Start G-code: "`G1 E-[retract_length] Z-[retract_lift] F{retract_speed[0] * 60}`"
results in the following inside G-code file:
**_G1 E-2 Z-0 F2400_**

#### Project File (.3MF) where problem occurs
N/A

rtyr, 2021-09-15:
Could you please try with 2.4.0-alpha1? I believe we already fixed this.
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1

cloud30000, 2021-09-15:
It is partially fixed in Version 2.4.0-alpha1+win64

When all Filament Override boxes are deselected, it uses the Printer settings.

However, when the Filament Override for Length is checked and set to 0 and all other overrides show as disabled, the "retract_length[0]" placeholder returns 0 but all other retraction_xxxxx[0] placeholders are reporting the last selected value.

In addition, when the Filament Override boxes are checked, those override values are being used for the retraction_xxxxx[0] placeholders; I do not know if this is intended, since the Filament Override settings already have their own filament_retraction_xxxxx placeholders. For instance, if I wanted to compare the Retraction speed of the Filament Override to the Retraction speed of the Printer Extruder and select the slower one due to an extruder limitation regarding retractions, I cannot access the Retraction speed of the Printer Extruder to compare against if the placeholder is overridden.

bubnikv, 2023-03-20:
Could you please provide an example 3mf project with which we may test your issue?

closed 2026-09-29 by prusaslicer-bot[bot]