prusa3d/PrusaSlicer #6956: Put in an option for RMB-hold to rotate, and LMB-drag to move objects

thanhtacles, 2021-09-15:
The controls are straight up unusably unintuitive and it wouldn't be much to simply add an option so that right mouse button hold is the one to rotate, and leftclick hold moves objects around. You never want your control to move objects be the same as the one that rotates the camera. That results in a ton of overlap between trying to rotate view and suddenly moving your objects. Nowhere does that. Copy Cura's controls, they're solid. You don't even have to change any defaults, just include this as an option it the camera controls.

ChaosBlades, 2021-09-18:
Cura, TinkerCAD, and Onshape use right click to rotate. Fusion 360 has an option to use TinkerCAD camera navigation. Would be nice if everything could use the same controls or at least have the option.

foreachthing, 2021-09-18:
And I want the same controls as in Blender. 
:stuck_out_tongue_closed_eyes:

natrius, 2022-03-20:
Kind of the same implementation like freecad would be awesome. It lets you choose from various styles and as i want to learn blender some time i chose blender right now. It would be awesome to get the same choices in PrusaSlicer.

[image]

RichardBronosky, 2022-08-26:
The way OpenSCAD allows you to navigate you view of 3D space is amazing when compared to Prusa. Here are a few features:
- Zoom at cursor
- Ability to orbit at any time (aka: Don't drag unselected object on click-hold-drag.)
- Orbit center of viewport
- Option to show 3D crosshairs at center of viewport

I find PrusaSlicer's 3D interface to be torturous. Please interview the community and make it better. I can't believe the people at Prusa are happy with it.

thanhtacles, 2022-08-26:
^Adding to Richard,
You don't even need to interview the community. Just make it a simple simple OPTION to let people pick the button so the controls can be normal and correct. This with the solve so many low hanging fruit issues with UI comfort and it's trivial to implement.

Absolutely no software should ever have orbit and drag object be the same lmm button. How someone could design something so infuriating and not have an option to do otherwise is beyond me.

codefoster, 2024-01-10:
Strongly agree that this should be configurable.

Big-Channie, 2024-12-23:
I'm constantly wrestling with the camera controls, this feature would be a game changer for me

closed 2026-09-29 by prusaslicer-bot[bot]