prusa3d/PrusaSlicer #6963: Small perimeters size to apply speed change appears in error

kevinmichaeldudley, 2021-09-16:
### Version
_Version of PrusaSlicer used goes here_ 2.3.0

_Use `About->About PrusaSlicer` for release versions_2.3.0+Win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_  Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_Prusa MK3S

### Behavior
* _Describe the problem_ Printing perimeters around holes is to fast
* _Steps needed to reproduce the problem_  
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_  Using Small Perimeters setting appears to apply to holes of 6.5mm DIAMETER or less NOT radius.
* Info says it should apply to 6.5 Radius <= but appears to be Diameter.
* _Actual Results_  
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ Yes and No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

I could not figure out how to send the project file *.3mf
In the 2 following images, VIEW = SPEED 
The holes in the part are 6.35 mm diameter.  
The Small Perimeter speed setting seems to only affect the hole size plus a little which is WAY less than 2 x 6.5mm = 13mm of setting per description.
Also it would be TREMENDOUSLY USEFUL to have a setting to choose the size of hole/radius since combinations of machines/materials/design etc.. can have varied requirements.

The LAST image red circles a section of the print which I do not understand why or how the Small  perimeters is changes this also??

[image]

Small Perimeters speed set to 50
[image]

Small Perimeters speed set to 5
[image]

Clasp, circled in RED why does small perimeters affect this?  It's OK it is helpful but unexpected.
[image]

rtyr, 2021-09-17:
I don't see any issue on your screenshots. Use print simulation (horizontal slider) to better see, that other perimeters are not small. 

[image]

> Using Small Perimeters setting appears to apply to holes of 6.5mm DIAMETER or less NOT radius.

No, it's a radius as stated in the tooltip.

> lso it would be TREMENDOUSLY USEFUL to have a setting to choose the size of hole/radius since combinations of machines/materials/design etc.. can have varied requirements.

https://github.com/prusa3d/PrusaSlicer/issues/1297

> I could not figure out how to send the project file *.3mf

It needs to be zipped first.

> Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)

rtyr, 2021-09-22:
Answered. No response. Closing.

closed 2021-09-22 by rtyr

MrBlackmidi, 2022-11-22:
this has happened to me aswell, have no idea why it happens 
[image]

MrBlackmidi, 2022-11-22:
with arachne btw

MrBlackmidi, 2022-11-22:
also the external perimeters of those little hexagons doesn't comply with external perimeter speed

rtyr, 2022-11-22:
Those are all small perimeters, so small perimeter speed is applied.