prusa3d/PrusaSlicer #6965: What is the default print speed of i3MK3S+ and can we change the overall speed in absolute value?

laptop-laptop, 2021-09-16:
### Version
2.4 alpha1

_Use `About->About PrusaSlicer` for release versions_
2.4 alpha1

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Prusa i3MK3S+

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_

What is the default print speed in mm/s for Prusa i3MK3S+? If I want to change the overall print speed to say 30mm/s, which parameter(s) should I change? Is there a way to enter the speed in absolute value in mm/s rather than in percentage?

I asked three Prusa Support agents but they could not provide a definite answer. One person just said it depends on a lot of factors so there is no default overall print speed. The other two just said that if I want to change the speed, I just go to Prusa Slicer and there are lots of speed parameters for me to change.  They could not tell which specific parameter nor which speed the % are based on. They also mentioned about going to the dial to change the speed. This does not answer my question. For example, if the default overall speed of i3MK3S+ is 100mm/s, then 30% means 30mm/s. However, if the default overall speed is 200mm/s, then 30% means 60 mm/s.

rtyr, 2021-09-17:
> What is the default print speed in mm/s for Prusa i3MK3S+?

It's not possible to give you one value, because such value simply doesn't exist. Speeds used in stock profiles can be considered as "default". As you probably know, every print profile uses different speeds for different features (for example 45mm/s for perimeters, 25mm/s for external perimeters, 80mm/s for infill, etc. - all set as absolute values.). 

> If I want to change the overall print speed to say 30mm/s, which parameter(s) should I change? Is there a way to enter the speed in absolute value in mm/s rather than in percentage?

`Print settings -> Speed -> Speed for print moves section`. For example like this, but it's really up to you which speeds do you want to change and why.

[image]

> Is there a way to enter the speed in absolute value in mm/s rather than in percentage?

Yes, we are actually using only absolute values, except for support material interface speed.

> They could not tell ........................... nor which speed the % are based on.

This information is provided in the tooltip (if you mean % in PrusaSlicer).

> They also mentioned about using the dial to change the speed.

When you rotate the LCD knob, you are overriding the speed directly in the firmware. The percentage is calculated over speeds defined in the g-code, so basically you will get for example 50% of 45mm/s for perimeters, 50% of 25mm/s for external perimeters, 50% of 80mm/s for infill (simplified example). This is probably not what you are looking for, so you have to set your prefered speeds in the slicer.

Please contact our tech-support with further support related questions.

fe60, 2021-09-17:
> This information is provided in the tooltip (if you mean % in PrusaSlicer).

[image]

closed 2021-09-18 by rtyr