prusa3d/PrusaSlicer #6966: [FR] First Layer no straight lines

wuast94, 2021-09-16:
### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Rat Rig V-Core 3 500 + Klipper

### Behavior
* _Describe the problem_
I have a huge model, its a big rectangle (400x400) and with PrusaSlicer it creates long lines for first layer (wich is correct) but on this size the problem is that bubbles take place no matter how much i play with settings. 
* _Expected Results_
A setting to make diffrent patterns on first layer would make it a lot easier on this, little rectangels or honycomb pattern for example. In my part are little rectangles and holes they stick just fine. So many little ones on solid areas should improve this
* _Actual Results_
* 
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
[Halterung.zip](https://github.com/prusa3d/PrusaSlicer/files/7180732/Halterung.zip)

_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-09-17:
The problem on the photo should be primarily fixed on the printer side (first layer calibration, leveling).

You can change bottom fill pattern.

[image]

dartrax, 2021-09-19:
In fact I know these artifacts. Reason should be too much material on first layer. Increase the distance between nozzle and build plate, like rtyr suggested.

wuast94, 2021-09-21:
> The problem on the photo should be primarily fixed on the printer side (first layer calibration, leveling).
> 
> You can change bottom fill pattern.
> 
> [image]

that make the trick, Hilbert Curve was the only one that hit my needs, a setting for length would be nice, its very fine for such a huge print 🙂

wuast94, 2021-09-21:
> In fact I know these artifacts. Reason should be too much material on first layer. Increase the distance between nozzle and build plate, like rtyr suggested.

i played so much with all this settings, and for "normal" sizes that works fine, on this huge print its not easy possible, adhesion was killing me, i ordered a glass plate and hope that i can go more precise with that, the flex sheet isnt "level" enough in my case

rtyr, 2021-09-21:
Maybe you could just use a raft (it was reworked in 2.4.0-alpha1).

> that make the trick

Can we close the issue then?

dartrax, 2021-09-21:
Ah, I understand. On large prints, adhesion can be difficult at the edges. Possible problems are decreased temperature at the edges of the build plate or increased gap between nozzle and bed because of bending rods or uneven print beds on large printers. Mesh bed levelling could be a solution.
Glass plate can have it's own challenges, depending on the filament... I've had better results with the pei sheet for PETG.

wuast94, 2021-09-21:
> Maybe you could just use a raft (it was reworked in 2.4.0-alpha1).
> 
> > that make the trick
> 
> Can we close the issue then?

i guess yes, but would like to see a setting to select the size of the Hilbertcurve.

closed 2021-09-21 by bubnikv