prusa3d/PrusaSlicer #6967: Print profile does not save a custom Top Interface Layers (quantity) value unless a dropdown is selected first

cloud30000, 2021-09-17:
### Version
Version 2.4.0-alpha1+win64

### Operating system type + version
Windows 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
When changing the "Top interface layers" value to one not in the dropdown list, that value is not saved to the profile after selecting the "Save current Print settings" button or when selecting "Save" in the warning dialog upon switching profiles without saving first.

However, if the  "Top interface layers" value is selected from one of the dropdown options first (and is different from the current setting), then that respective selection can be deleted and the custom value can be entered in it's place. At that point, selecting Save does save it to the profile.

After changing the value to a dropdown and then to 4, the value saved as expected. Then changing the value to 10 without using the dropdown and saving, the value displayed as 4 again when changing profiles and returning to the modified profile. Closing and reopening PrusaSlicer also continued to display it as 4. Then, changing it to a dropdown first, and replacing the dropdown text with 10 and saving it correctly displayed the value as 10 when switching profiles and returning as well as when restarting PrusaSlicer.

It seems as though PrusaSlicer does not correctly recognize that the field has changed unless it is changed to the dropdown, and if it doesn't recognize that it changed then it does not try to save that ("unmodified") field.

#### Project File (.3MF) where problem occurs
N/A (Observed this when trying to save custom versions of the included Prusa MK3S+ without MMU 0.4 nozzle profiles, and all of them were affected in the same manner.

rtyr, 2021-09-20:
Is it 100% reproducible on your PC? I have a problem with reproducing that issue.

cloud30000, 2021-09-20:
When observed, it happened every time. However, now (a few days later) I am not able to reproduce it and I do not recall any special circumstances that would have led to the application being in a special state. I was going through multiple profiles to update the Top Interface Layers of each of them, but that does not seem to reproduce the issue at this moment.

closed 2026-09-29 by prusaslicer-bot[bot]