prusa3d/PrusaSlicer #6971: Print settings not applied to all copies if one copy rotated

dartrax, 2021-09-18:
### Version
2.4.0 alpha 1

### Operating system type + version
Win 10

### Behavior
* _Describe the problem_
If I have a copy of a model and rotated that by 90° Z-Axis, changes in print settings like changing Infill type or perimeter count will apply only to one of the models. I need to toggle the visibility of the other copy to have the settings applied there, or save and open the project will also workaround this issue.

* _Steps needed to reproduce the problem_
Add a model, duplicate it by clicking the + button, rotate the copy by 90° Z-Axis, then change the perimeter count
* _Expected Results_
New setting will be applied to both models
* _Actual Results_
Setting is applied only to one model

_Is this a new feature request?_
No

troypouliot, 2021-09-20:
I just ran into this same issue now. 6 copies of a model, some rotated, some not. 5 of the 6 would not take changes, then I set all models as separate objects. Now only 1 of the 6 will not take print changes.

bubnikv, 2021-09-20:
Fixed with f484953a5a1fecd878242ca8d2f5175151b81678

closed 2021-09-20 by bubnikv