prusa3d/PrusaSlicer #6972: Error compiling PrusaSlicer 2.4.0A1 on Linux 19.3

Area5142, 2021-09-18:
### Version
_version_2.4.0-alpha1-138-g9d7549e66_

### Operating system type + version
_Linux Mint 19.3 x64_

### 3D printer brand / version + firmware version (if known)
_NA_

### Behavior
* _Error compiling 3mf.cpp. Missing standard header \<charconv\>:_
<pre><font color="#4E9A06">Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Format/3mf.cpp.o</font>
In file included from <b>/home/.../src/PrusaSlicer/src/libslic3r/Format/../GCode.hpp:6:0</b>,
                 from <b>/home/.../src/PrusaSlicer/src/libslic3r/Format/3mf.cpp:6</b>:
<b>/home/.../src/PrusaSlicer/src/libslic3r/Format/../GCodeWriter.hpp:6:10:</b> <font color="#EF2929"><b>fatal error: </b></font>charconv: No such file or directory
 #include <font color="#EF2929"><b>&lt;charconv&gt;</b></font>
          <font color="#EF2929"><b>^~~~~~~~~~</b></font>
compilation terminated.
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:508: recipe for target &apos;src/libslic3r/CMakeFiles/libslic3r.dir/Format/3mf.cpp.o&apos; failed
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/Format/3mf.cpp.o] Error 1
CMakeFiles/Makefile2:1080: recipe for target &apos;src/libslic3r/CMakeFiles/libslic3r.dir/all&apos; failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:145: recipe for target &apos;all&apos; failed
make: *** [all] Error 2
</pre>

* _Installed GCC version_
```
gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
```
The use of \<charconv\> is a recent change to the source as the 2.4.0-alpha1-15-gb69d03ad4 source compiled without problems.

Searching the web it looks like \<charconv\> standard header is only supported by the newest gcc/g++ versions.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_NA_

hejllukas, 2021-09-20:
_std::to_chars_ and _std::from_chars_ from _<charconv>_ should be supported from GCC 8 for integer numbers (We use GCC 8.3 for Raspberri Pi builds, and it supports std::to_chars for integer numbers). 
Floating-point numbers in _std::to_chars_ are supported since the newest GCC 11, but we use function _std::to_chars_ just for integers, so it should be sufficient to use GCC 8.
We decided to use _std::to_chars_ because G-Code export was very time-consuming. We did some measurements of performance and found out that _std::stringstream_ is very slow compared to the _std::to_chars_. So we refactored some parts of G-Code export and replaced _std::stringstream_ with a custom function that uses _std::to_chars_.
So I would recommend using a newer version of GCC for building PrusaSlicer.

Area5142, 2021-09-21:
Installing gcc 8.x and g++ 8.x on Linux Mint 19.3 (Ubuntu 18.04) solves the include problem and makes the current code (head) compile again:

```
# Install gcc v8 and g++ v8
sudo apt install gcc-8 -y
sudo apt install g++-8 -y

# Install compiler as new default alternative
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 700 --slave /usr/bin/g++ g++ /usr/bin/g++-7
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 800 --slave /usr/bin/g++ g++ /usr/bin/g++-8

```
It is recommended to install the newer compiler side-by-side with the default version 7.x. Using _update-alternatives_ the new version can be prioritized before the older version and no other change is needed on the system.

Area5142, 2021-09-21:
I am closing this issue as the compiler problem is solved using a GCC /G++ version 8.x or newer on the Linux platform.

closed 2021-09-21 by Area5142