prusa3d/PrusaSlicer #6976: [FEATURE REQUEST] Easily detachable brim

uzytkownik, 2021-09-19:
### Version
2.3.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3+

### Behavior
I had a problem with warping when printing long thin object. Skirt didn't helped and when I tried to use brim it was very hard to detach.

I created manually brim using boxes around the object and it worked - the supports where easily detachable and it get better adhesion. 

[image]

This is much more wasteful than what could be done automatically - I needed to eye the sizes so they are larger than necessary, put everything in single object etc. Would it be possible to have an option of printing brim the same way as supports to have it easily detachable?

rtyr, 2021-09-19:
We added a new parameter "Brim separation gap", allowing the user to fine tune the brim offset from the object. It is already implemented in PrusaSlicer 2.4.0-alpha1.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1

[image]

bubnikv, 2021-09-22:
Implemented in 2.4, closing.

closed 2021-09-22 by bubnikv