prusa3d/PrusaSlicer #6978: PrusaSlicer v2.3.3+win64 does not display the name of .3MF project files in the windows title bar if they are opened by double-clicking.

MucaroBoricua, 2021-09-19:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK2.5S

### Behavior
* Double-clicking a .3MF file opens it up in PrusaSlicer but the window title bar does not show the name of the .3MF file like when the same file is opened with the "File/Open Project..." menu option.

* To reproduce the issue, just double-click a .3MF file and notice that the PrusaSlicer window title bar just displays "PrusaSlicer-2.3.3 based on Slic3r".

* The expected result is to see the name of the .3MF file like when it's opened with the "File/Open Project..." menu option.  For example, double-clicking the file "PCB Holder - Posts.3MF " should set the window title bar to "PCB Holder - Posts - PrusaSlicer-2.3.3 based on Slic3r".
[image]

* The actual result is a title bar displaying "PrusaSlicer-2.3.3 based on Slic3r".
[image]

### Is this a new feature request?
No, this a minor, but very annoying, bug.
**This bug was originally reported in issues #6473 and #6505 but both issues were closed without actually fixing the bug.**

rtyr, 2021-09-19:
Already fixed in 2.4.0-alpha1, as stated here https://github.com/prusa3d/PrusaSlicer/issues/6473#issuecomment-853897625.

You are using older PrusaSlicer 2.3.3.

rtyr, 2021-09-19:
https://github.com/prusa3d/PrusaSlicer/issues/6505#issuecomment-922525254. Closing.

closed 2021-09-19 by rtyr