prusa3d/PrusaSlicer #6979: PrusaSlicer 2.4.0-alpha1 MMU Painting Choose Brim Color/Extruder

killerb77, 2021-09-19:
### Version
PrusaSlicer 2.4.0-alpha1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5S with MMU2S

### Behavior
I am testing out the MMU painting feature.  When I add a brim it doesn't know which extruder/color to choose.  As per the screen capture the brim should be black/extruder 4.  It chose white/extruder 1.  Not sure if you can make it choose the outer most color/extruder or add a feature to choose the extruder/color for the brim?

screen capture and project file attached.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save
[C3DC Maker Coin_MMU.zip](https://github.com/prusa3d/PrusaSlicer/files/7192343/C3DC.Maker.Coin_MMU.zip)
[image]

singultus, 2021-12-26:
Same here. This is a problem especially for mmu prints, as the residue of a peeled off raft discolors the print itself. I hoped this will be fixed in 2.4 with individual brim. However, I failed to force the same printhead for brim and associated object. Are we missing the proper way to do it?

rob-miller, 2024-10-26:
#10491 ?

closed 2026-09-29 by prusaslicer-bot[bot]