prusa3d/PrusaSlicer #6981: Not print on my Sidewinder X1

yvesn007, 2021-09-20:
### Version
_Version of PrusaSlicer used goes here_2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Win 10 64 bits
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Artillery Sidewinder V4 origin

### Behavior
* _Describe the problem_ Heah place for print and no print.
* _Steps needed to reproduce the problem_ Juste tried print part
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-09-21:
> Heah place for print and no print.

Could you please provide more detailed description and project 3MF file?

yvesn007, 2021-09-22:
I think it’s very simple. But i’m new. I make the same in Cura and no problem???

Before the last update he work very well.

Thanks. 

Envoyé à partir de Courrier pour Windows

De : rtyr
Envoyé le :21 septembre 2021 00:31
À : prusa3d/PrusaSlicer
Cc : yvesn007; Author
Objet :Re: [prusa3d/PrusaSlicer] Not print on my Sidewinder X1 (#6981)

Heah place for print and no print.
Could you please provide more detailed description and project 3MF file?

rtyr, 2021-09-22:
Please provide 3MF project file (File - Save project as..) or at least g-code file and some description of your problem (what exactly is happening when you start the print).

yvesn007, 2021-09-22:
Then printer make a purge line on X axe after buse go to start print. The printer Begin for piece first line on 2 or 3 cm and at the same time I hear the engine stopped gradually and quickly.

Do you have enough information and received the correct attached files?

Thanks.

YN

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

rtyr, 2021-09-22:
Please attach the files directly via github web interface. It is unfortunately not possible to attach files by e-mail.

yvesn007, 2021-09-23:
Ok now it's working. I think it bad configuration with modification speed limit on the printer.

Now ok

Do you want file anyway 🧐

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

rtyr, 2021-09-23:
No, I just wanted to check your settings.

> Ok now it's working.

Good. Closing.

closed 2021-09-23 by rtyr