prusa3d/PrusaSlicer #6982: Feature Request: Surface Mode from Cura

TheRooster13, 2021-09-20:
Cura has a Special Mode called Surface Mode. I'd love to get the same thing in PrusaSlicer. It will help me immensely with some models I'm trying to print.

One addition to Cura's Surface Mode that I would request would be the ability to alternate starting locations from one layer to the next, so if the print is all one surface, there wouldn't need to be travel when stopping one layer and starting the next. So the extrusions would go back and forth up the model. So similar to the "nearest" setting on seam definition.

RickBoerma, 2021-09-23:
There is already the hollowing function in the SLA mode. This function would be great for FFF too, such that you can print a surface model with a specified wall thickness (and infill if required for stiffness). 

If this option can also be extended by flipping the direction/side of the print then the print can also be used as mould.

TheRooster13, 2021-09-23:
Hollowing is not what I'm looking for. I can easily hollow stl files today with various programs, but the results are not a single extrusion at the surface for arbitrary meshes.

Also, you need a closed mesh to hollow. Surface mode doesn't require a closed mesh.

intra-au, 2021-11-09:
+1 this is the only reason for me not to use PrusaSlicer, if Surface Mode can be added - only slicer I'll need will be Prusa.

lukasmatena, 2021-11-09:
@TheRooster13 Could you please expand on how the Cura's surface mode differs from settings Top and Bottom layers to zero and infill density to 0 % ? Thanks.

TheRooster13, 2021-11-09:
@lukasmatena Surface mode doesn't require the path of each layer to be a complete circuit.

An STL could be created that isn't watertight. For example, a vertical cylinder with a hole in the side. The sliced layers would start as complete circles, but when the hole in the side is reach, the slices would resemble the letter C instead of O. Then above the hole, the slice would be a complete circle again.

Surface mode would be able to print this non-watertight model just fine. Regular slicing would need to fill the hole first so that every slice makes a complete circuit.

Surface mode has no concept of inside and outside the model. There are just surfaces. Regular slicing requires a complete circuit for every slice because there needs to be an inside.

Hope that is clear. I don't have time to make pretty pictures right now. :)

neophyl, 2021-11-09:
It basically allows printing of non manifold volumes.

bubnikv, 2021-12-21:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/6552

TheRooster13, 2021-12-21:
I see the "awaiting response" tag is still applied. @lukasmatena, do you need more information to answer your question?

One more difference, I'm pretty sure surface mode just extrudes centered on the surface because there is no concept of inside or outside in surface mode.

IkonOne, 2022-12-22:
Here is why Surface Mode is not a stupid hack.

I work for a lab doing research on 3D printing bone tissue for regrowth due to catastrophic failures.  The material used is a UV-cured material that is too thick for SLA.  So we put it in a syringe on a 3-axis printer with a UV ring light around the head of the syringe to cure the filament on deposition.

It's a hack.  But it works.  But only barely.

The problems arise when there are z-hops.  Due to its viscosity, the material leaks from the syringe.  We have to be very efficient with limiting z-hops to ensure that the bone scaffolds are printed to within an acceptable level of quality.  Cura's Surface Mode is the only way we have been able to do that with the topologies that we are printing.

[Here is a PR](https://github.com/Ultimaker/CuraEngine/pull/1536) I submitted to fix some z-hop issues with Cura's Surface Mode.  While what you see in the images in that PR is gyroid, realize that there are [many other triply periodic minimal surfaces](https://www.google.com/search?rlz=1C1ONGR_enUS1035US1035&source=univ&tbm=isch&q=triply+periodic+minimal+surface&fir=eE7VdQ4YbtTM9M%252CakOU-wmSGAzuJM%252C_%253BP-9iIErt6Yz_rM%252ClqqQp_7_5SQc1M%252C_%253BMpaPclOPyskloM%252Cle8-mZh-4FHSWM%252C_%253B_xAyjR8OCIrbGM%252C5Hlouv4ZXhgtwM%252C_%253BMOBcun_4BXkX2M%252CuvypGz4rE40_QM%252C_%253ByIy68aCh0wEA3M%252CetoVIB8HstPDLM%252C_&usg=AI4_-kTdXr1l3VmmEbo6lmNbejNTOS8IFw&sa=X&ved=2ahUKEwjg2PfRoY78AhWtAjQIHaLxDZAQ420oAHoECBAQBQ&biw=1536&bih=810&dpr=1.25).  Also recognize that there is a growing body of research that shows that [continuous, aligned filament deposition 'roads' provide significant gains in structural integrity](https://dl.acm.org/doi/pdf/10.1145/3414685.3417834).  While it is true that you can often [shell ](https://github.com/fogleman/sdf#shell)these fancy surfaces, even doing so does not result in continuous, uniform shells, and subsequently filament deposition paths.

Here is a shelled gyroid for an example.  The other TPMSs you can see in the images above behave similarly...
[image]

By adding a Surface Mode feature you enable the printing of novel, non-manifold surfaces with a high degree of control over the filament deposition paths.  You enable academic research into 3D printed tissue scaffolds.  You create an entirely new direction for your advanced users to learn and explore with a very low barrier to entry.

Here is an example of the type of non-manifold surfaces that we need to print for research purposes:
[image]

ChelA74, 2024-03-29:
Hello
I'm also interested by CURA Surface mode équivalent in PRUSA for RC models
CURA Surface mode is genius !
Any solution ?
Thanks

TheRooster13, 2024-03-29:
Unfortunately, no.

closed 2026-09-29 by prusaslicer-bot[bot]