prusa3d/PrusaSlicer #6983: Bug : New item / opened project doesn't show up in preview tab

MathieuGuennaoui, 2021-09-20:
### Version

Version 2.4.0-alpha1

### Operating system type + version

macOS Big Sur 11.5.2 on a MBP 15" (2016).

### 3D printer brand / version + firmware version (if known)

Prusa i3Mk3S (3.10.0) + MMU2S (1.0.6), almost stock - slight hardware upgrades.

### Behavior

Background processing is activated.
When adding a new .stl or .3mf (either with the button (+) or by drag'n'dropping) and moving from editor view to preview, it doesn't show at all, despite showing the export button ready. A listing of parameters or gcode appears instead.
I tried to untick the background processing and slice manually, but the same behavior happens.

To work around, all I have to do is change a parameter (ie: Perimeters 2 > 3) and the object shows up. Back to normal behavior after that, even if you add new items.

#### Project File (.3MF) where problem occurs
[image]
[image]
[Forme-Case.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7198517/Forme-Case.3mf.zip)

dartrax, 2021-09-20:
Same for me, I also have this problem.

Area5142, 2021-09-21:
On Linux Mint 19.3 x64 I have the same behavior.
Placing a model on the build plate and switching to preview shows empty preview. Toggling the brim on/off forces the preview to be shown.
(Background processing is on)

I reported a similar issue in 2.3.0A0 #4719

bubnikv, 2021-09-21:
It happens just once after application start if you happen to switch to the G-code viewer after the first object was fully sliced. So the issue only happens if the G-code was already fully generated and after that the user switched to the G-code viewer for the first time. It looks to me really as if the G-code viewer OpenGL context or whatever was not ready yet when the "G-code generated" event was sent to the G-code viewer, thus the event is ignored in that case.

@enricoturri1966 would you please take care of it? Thanks.

bubnikv, 2021-09-25:
Fixed in PrusaSlicer 2.4.0-alpha1

closed 2021-09-25 by bubnikv