prusa3d/PrusaSlicer #6984: [Festure Request] Fan speed scaling on a per machine basis

Schild0r, 2021-09-20:
I love the 'modularity' of settings in prusa slicer because you can mix and match different quality presets for different filaments on different machines. However using one Filament profile on different machines is only possible if their respective part cooling fans behave similarly.

One example: I have one small machine that has dual 30mm blower fans for part cooling, they do the job but they need quite some duty cycle to be effective while another big machine I have uses a 5015 blower that can literally cool down my heated bed at 100% duty cycle.
So when I want to print with the same filament profile on both machines either one of them won't cool enough resulting in curling or one will cool too much resulting in weak prints. Or I have to adjust the fan speed in the filament profile every time. 

My suggestion:
Implement a fan scale for machines in the printer settings tab that does not work as a hard cap (all fan speeds below are allowed, all speeds above are set to this cap = limit) but more as a upper scaling limit or a factor if you will. For example if you set the fan scale of the very powerfull fan to 70% and the filament profile requests for a 50% fan speed it will result in the fan being set to 0.7*0.5=35% speed. While the fan scale on the weak fan can be set to 100% and that will result in the fan being set to 50% speed with the same filament profile.
To add to this you could also implement an (optional) parameter for the minimum allowed absolute fan speed, so that the scale won't result in the calculated fan speed being set too low so it won't spin up.

I am also thankfull if one would tell me that there is already a workaround to this, that does not require a postrprocessing script.

Keep up the great work!

dartrax, 2021-09-20:
I also thought about that.
But then, I thought I should better tune the firmware settings of my printers so that the fan speeds match each other. For example, if you use Marlin, you could set
```
#define FAN_MIN_PWM 50
#define FAN_MAX_PWM 128
```
(change the values according to your needs) in your Configuration_adv.h file and recompile your firmware.

Schild0r, 2021-09-20:
Well I don't use Marlin but RRF (on both mentioned printers) and afaik I can only set a max speed but it won't scale it down from that value.

murk-sy, 2021-09-21:
While this is something I would like to have (since it improves print consistency), it would also require you to have *some* sort of a hard value to base everything off of. Just going off of fan's parameters (such as m^3/h) could be a start, but actual cooling performance is affected by the cooling duct shape, duct exhaust size, ambient temperature, nozzle temperature, fan type, and a bunch of other stuff that I barely know the name of.

Since most consumer materials and applications are not too demanding controlling all the factors is unrealistic, this isn't high on the to-do list. An option would be to calibrate fan speed using nozzle temperature drop to calculate *something*. At 0, 20, 40% etc, heat the nozzle to 200°C and then measure how long it takes to cool 25°C. Although a relatively easy process, would that even be a useful measurement? Who knows.

I like using as little cooling as possible for best layer adhesion, but any ambient temperature changes affect enclosure temperatures too. Ideally for a business, you have printers in a climate controlled environment with just a bunch of the same model to ensure consistency.

For now, you can create multiple profiles of the same materials and use machine compatibility settings, though that can be a bit of a pain to manage.

dartrax, 2021-09-21:
> Well I don't use Marlin but RRF (on both mentioned printers) and afaik I can only set a max speed but it won't scale it down from that value.

May be this helps, may be not:
https://github.com/Duet3D/RepRapFirmware/pull/258
Is this undocumented?

rtyr, 2021-09-21:
Why just not create a separate filament profile and set its dependecies so it will be visible only for the printer it's meant for?

dartrax, 2021-09-21:
Doesn‘t that mean redundancy of the other filament values with all the negative implications that has?

rtyr, 2021-09-21:
Possibly, if we are talking about two very similar printers which have just different print fan... But even in that case, there may be still a need to change other parameters as well (like minimal time per layer, full fan speed at layer, ...). Anyway, it is quite easy solution which is already available.

Schild0r, 2021-09-21:
> Why just not create a separate filament profile and set its dependecies so it will be visible only for the printer it's meant for?

That is the only option for now I guess if you just consider the slicer side of things however if you make changes you have to always change both profiles as long as there is no derivative profiles that have (the chosen) parameters linked to each other.

dartrax, 2021-09-21:
Yes, as long as you exclude postprocessing scripts ;-)
Not that I know how to do that, but I think that should be a possibility without having redundant filament parameters

Schild0r, 2021-09-21:
> May be this helps, may be not:
> [Duet3D/RepRapFirmware#258](https://github.com/Duet3D/RepRapFirmware/pull/258)
> Is this undocumented?

Just read into this and this is basically the exact thing I wanted to have. If this still works this would solve my problem I think: I could just set the maximum speed and enable scaling either in the fimware directly OR in the slicer.
However the current [wiki entry of M106 for RRF](https://duet3d.dozuki.com/Wiki/M106) does not list the W parameter (neither does M950) so either this is a hidden feature that is only documented in the forum or it was axed at some point between RRF2.03 and RRF3.3
Once I come home I will try if it still works.

So if there is firmware support for this feature in the three major firmwares Marlin, RRF and Klipper (I heard that one is all the rage right now ;) ) this FR can be closed but if one or two FWs don't support it, it might be worth adding to Prusaslicer IMO. Maybe some Klipper person can confirm or deny whether this is already supported for them.

closed 2021-09-22 by Schild0r