prusa3d/PrusaSlicer #6985: Bug / Feature Request: Can't right-click vertical layer bar to change color of bottom section of color-changed print

MNeMoNiCuZ, 2021-09-20:
### Version
2.3.3

### Operating system type + version
Windows 7

### Behavior
When you add a color change by pressing the +-button next to the vertical line, it adds a new color change and it gets a random color.
You can change this color by right-clicking on this bar (called "Edit current color"). But this doesn't work on the bottom-most color. This is not coherent in it's design and feels like a bug.

In order to change the color of the first layer, you have to click on the color swatch of the Filament drop-down menu, which is much less intuitive. It doesn't look clickable, and it doesn't have a mouse-over prompt, and even when you change it, it shows 2 colors in it's palette:

[image]

Mostly it's not intuitive because it's part of a drop-down menu. You don't expect this result when clicking it.

_Is this a new feature request?_
A bit, maybe?

rtyr, 2021-09-21:
> But this doesn't work on the bottom-most color.

Bottom-most color is not related to any color change, it uses filament color. Clicking on filament color to change the color seems quite logical, but I can understand it may be not 100% intuitive for everyone. It's main purpose was to change the color for multi-material prints (colors are selected for more filament profiles).

MNeMoNiCuZ, 2021-09-21:
True, it's not related to a color change, but it is representing a color, the chosen filament color. But all the other colored segments on this part of the UI has a function, and this part doesn't. It becomes an exception rather than a rule. That feels unintuitive and it doesn't match the other features in this part of the UI.

MNeMoNiCuZ, 2026-09-04:
This bug is still here, and now there's no "+"-button to understand how to add new colors/a pause. The interface is worse here currently. The vertical bar needs visual improvements.