prusa3d/PrusaSlicer #6987: 3mf files tend to be placed outside the print bed area.

thanhtacles, 2021-09-21:
### Version
2.3.3+
There's a common problem among all 3mf files wherein I believe the position is memorized and thus whenever you plop them onto the slicer bed, it'll always try to reflect the CAD even if it's outside the area. this is a common problem considering almost all community/cad stuff is done at 0,0 origin, so anything that isnt in the positive x and y will always by default be left outside the print bed.

I think a simple solution would be to have prusaslicer intelligently recognize where an individual object should be placed, and even consider collisions of files clipping through each other

rtyr, 2021-09-21:
This should be already fixed in PrusaSlicer 2.4.0-alpha1.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1

bubnikv, 2021-09-22:
fixed in 2.4, closing.

closed 2021-09-22 by bubnikv