prusa3d/PrusaSlicer #6988: Paint on Fuzzy Skin

mann1861, 2021-09-21:
The fuzzy skin in 2.4.0alpha is working really well for me especially with modifiers to enable in specific locations however for some models it is quite tedious to make a separate stl to enable the fuzzy skin in the correct locations. A good implementation would be similar to the new paint on colours in 2.4.0alpha to be able to select the surfaces where fuzzy skin is required.

Here's an example where I used modifiers to get the finish I wanted but painting would have been easier
[image]

dartrax, 2021-09-21:
Duplicate of #6881

pmottram, 2022-01-09:
A basic version would extrude the painted faces normal to the mesh by 2x extrusion width in each direction (or just into the model if it knows the difference), then create this as a new modifer region with fuzzy skin. Currently I'm doing this manually in CAD (see https://www.prusaprinters.org/prints/115317-pi-zero-tv-case) but would be great to get this as a core feature as it would be so much easier to apply to any model.

reddawgesq, 2022-03-14:
+1 on this, much needed.

lukasmatena, 2022-03-14:
> Better, perhaps, would be to implement it as an "Add Modifier" or "Add Part" feature which you could shape and place more precisely and assign the fuzzy attribute to it. 

Maybe it is too early in the morning, but isn't this exactly how it works now?
[image]

reddawgesq, 2022-03-14:
Don't bother me now, I'm too busy re-inventing the wheel!

Snuff1eupagus, 2023-02-22:
> > Better, perhaps, would be to implement it as an "Add Modifier" or "Add Part" feature which you could shape and place more precisely and assign the fuzzy attribute to it.
> 
> Maybe it is too early in the morning, but isn't this exactly how it works now? [image]

Yes it is, and it's a wonderful method, that for some reason people can't seem to be bothered to learn.

zandr, 2023-02-22:
Yes, modifiers work today. But we have painting and Smart Fill already for supports and colors, and It would be a huge time saver to be able to use smart fill to apply fuzzy skin. Go back and read the original request in this issue, or any of the half-dozen dupes against it, not the one guy who didn't get it.

mann1861, 2023-02-22:
> How do you think I got here?
> I don't see the need for paint on fuzzy skins, the current method works
> wonderfully, I use it from time to time, and if your setting things up
> properly in CADD before importing your model, it's easy peasy. Sometimes
> when working with someone else's model it can take a bit more time, but
> there is a way, it's not bad, it's just not instant.
> That's my opinion, I've seen many of these, and other requests from people
> that don't bother to learn what can be done before making a request. That
> one prompted me to respond. I personally would rather see the limited
> resources of the development team used to correct core functionality in
> slicer, before seeing some limited implementation of a paint on modifier.

As the original author of the request I can assure you I am aware that
modifiers can be used. I even mentioned it and showed a print that used
modifiers to achieve the fuzzy skin result. That doesn't make my request
invalid. To use modifiers to make a surface fuzzy skin takes several steps
in cad and if complex it takes a while. Simply using the smart select the
same way as the existing paint on colours or support it would be much
simpler.

Your argument can be said with paint on colours in the slicer as that could
be done by splitting th object in cad but you do not seem to be arguing
that feature is unnecessary

Snuff1eupagus, 2023-02-22:
@mann1861 

Your ability was never in question, I read all that you wrote, and I use the same method successfully. I commented on the remark of another user, that clearly did not know, and that apparently caused @zandr to respond. That's how it goes. 

Using the same method, and applying it is not causing me distress even with very complicated models, yes it takes some time in some situations, I agree. However I try to weigh that when looking at other needs, and that brings me to my opinion on the subject. Frankly, I probably wouldn't have posted further here had my comment not drawn that response.

The bottom line is we don't get to decide the developers management does. All we can do is offer our opinions and make our requests.

Alexx40in, 2023-07-26:
oh how I'm waiting for a bug fix in this function. Painting would be perfect, but modifiers don't help now if the outer walls are thin, for example 2 perimeters. The bug with the modifier not working is terrible

exzenter, 2023-08-03:
https://www.reddit.com/r/3Dprinting/comments/12r0ccf/found_a_neat_trick_to_paint_on_fuzzy_skin_using/

Thats a workaround that helped me out...

zuidwijk, 2023-09-11:
I'd vote for this feature too, to paint support the way you can paint color. Printed lately this [boo bear](https://www.printables.com/model/567289-boo-bear) and was not able to apply fuzzy skin only on the arms and legs. 

Painting it would be sooooo much easier!

rbmuir, 2023-09-27:
Bump. Having a better method to enforce fuzzy skin only on certain surfaces is needed. Was trying to print a speaker box and it always textures the inside to, adding a ton of extra print time. Using shape modifiers is broken and doesn't work. It adds extra perimeters in random spots. You can kinda hack it a bit using a "IIIIIIIIIII" text modifier with the character spacing turned way down. If you try to do a whole model like that though the slicer will crash.

exzenter, 2023-11-16:
push!

Alexx40in, 2023-11-21:
up!

exeljb, 2023-12-17:
I need this to be a thing as well.  If the outside walls actually detected outside walls and was not foiled by holes in an object then this wouldn't be a problem for me.  It's also a very good idea to be able to have the option to paint on/smart fill the specific surfaces people might want the fuzzy skin on without the need of fabricating multiple special modifier objects to get the job done.  Either way, both of these issues should be addressed to get it working right and easy for people to use as they need.

[image]

ens-gijs, 2024-03-30:
Paint on fuzzy skin would address my needs - but so would fixing the (probably very hard to fix but ultimately probably the result of an ancient lazy decision) issue with all modifier bounds adding perimeters.

This is so frustrating - the injection of extra perimeters with all (or many?) modifiers is just horrible.

gaines, 2024-04-25:
+1 for paint-on being of value even if they fix the modifier perimeter issue.

zjohnson16, 2024-05-22:
+1 for paint-on fuzzy skin

NateBatchelder, 2024-06-01:
+1 bump

Twostroke500, 2024-06-18:
+1 i would love to see paint in modifiers in prusaslicer, especially with support for fuzzy skin.

I gave some of the workarounds a go with a complex grip part. I had to model a seperate freeform body to apply fuzzy skin to specific suraces of the part. Primitive geometry led to fuzzy skin on the inner mounting surfaces of the part, which is bad.
In summary this was roughly a 2h job which could be a 2min job with paint on modifiers.

[image]

Also the workarounds with modifier bodies does not solve the problem of fuzzy skin and/or additional walls inside of the part, which increases print time

[image]

jwaes, 2024-07-03:
Selecting distinct surfaces or paint surfaces to apply the fuzzy skin to would be such a game changer.

Michl59, 2024-08-18:
+1 for paint on fuzzy, especially when only needing one side of an object or just a face/surface of a part, without adding additional perimeters

lukasmatena, 2024-11-21:
Implemented in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena