prusa3d/PrusaSlicer #6992: PrusaSlicer gets stuck in an endless "Warning" loop if you set the "Avoid crossing perimeters - Max detour length" value to be significant (e.g. 10) without using a "%" sign

cloud30000, 2021-09-22:
### Version
Version 2.4.0-alpha1+win64

### Operating system type + version
Windows 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
PrusaSlicer gets stuck in a "Warning" loop if you set the "Avoid crossing perimeters - Max detour length" value to be significant (e.g. 10) without using a "%" sign.
Selecting the "Yes", "No", or "x" buttons all just cause the Warning dialog to instantly regenerate.

There does not seem to be a workaround to exit the loop, as even positioning the buttons directly over the length text entry field does not return to the field for editing, just regenerating the Warning dialog again instead. PrusaSlicer must be closed from Task Manager to get rid of Warning dialog.

It appears as though the loop is detecting if the entry field is not currently the focus, which it cannot be anymore once the warning occurs, instead of detecting if exiting the field for the first time.

dartrax, 2021-10-31:
Could not reproduce in 2.4.0 beta 1. Fixed?

cloud30000, 2021-11-01:
I figured out what caused it; after entering a significant value in the field (such as 10), switch to a different window/application in Windows. The warning dialog pops up in the PrusaSlicer app while it is in the background, at which point refocusing on the PrusaSlicer application leaves it in the undismissable Warning loop.

FidelCapo, 2021-12-02:
This issue should be fixed.
Closing.

cloud30000, 2021-12-03:
Please add the commit that addresses this issue, or else state that it could not be reproduced and what version you attempted to reproduce it.

cloud30000, 2021-12-03:
Reproduced in Beta 2; opening a new issue since I cannot reopen this one.

lukasmatena, 2021-12-03:
Thanks, reproduced.  Reopening.

bubnikv, 2021-12-04:
@cloud30000 
3622f06bed6175a337d2e20b5dc6ae9d770cb3cf seems to fix this issue. Would you please confirm?

cloud30000, 2021-12-04:
Downloading and installing Visual Studio now

cloud30000, 2021-12-04:
@bubnikv 
Can't compile; Microsoft updated the VS Community edition they are providing to 2022, but the PrusaSlicer GIT Repository CMake files keep giving me an error saying they are incompatible with this version of "MSVC", even after I deleted the CMAKE cache and tried using "cmake .. -G "Visual Studio 17 2022" -DDESTDIR="C:\local\destdir-custom"" from the VS "x64 Native Tools Command Prompt for VS 2022".

Is there a compiled Windows 64-bit version of this commit available to test with?

bubnikv, 2021-12-04:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta2+96-win64-g303de63a9-202112040813.zip
You may want to enter user name "prusaslicer" password "slicer" if a login window pops up.

dartrax, 2021-12-04:
Could not reproduce any more with 2.4.0 beta 3.

bubnikv, 2021-12-04:
Cool, thanks.
Closing

closed 2021-12-04 by bubnikv