prusa3d/PrusaSlicer #6993: Small hieght combobox in "Settings tab as menu items"

Shama71, 2021-09-22:
### Version
2.4 alpha1

### Operating system type + version
Windows 11 64bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Small hieght combobox in "Settings tab as menu items" mode.

_Is this a new feature request?_
[image]

#### Project File (.3MF) where problem occurs

dartrax, 2021-09-22:
I can confirm on Win 10 64 bit.
Display scaling is 125% if that's important.
[image]

bubnikv, 2021-09-25:
Fixed in PrusaSlicer 2.4.0-alpha1

Shama71, 2021-09-25:
Problem still exists.
[image]

dartrax, 2021-09-25:
Interesting, on Win 10 x64 it seems to be fixed.

[image]