prusa3d/PrusaSlicer #6997: Version 2.4.0 Unexpected change in print speed every few layers

tuxmobil, 2021-09-22:
### Version
2.4.0-alpha1

### Operating system type + version
Win 10 64-Bit

### 3D printer brand / version + firmware version (if known)
MKS 2.5S

### Behavior
The speed the main object is printed with changes to a (very) slow speed every few layers. This is not expected. 
See the attached .3mf.

[image]

_Is this a new feature request?_ No.

#### Project File (.3MF) where problem occurs
[Rahmen.zip](https://github.com/prusa3d/PrusaSlicer/files/7211840/Rahmen.zip)

rtyr, 2021-09-22:
Most likely related to fix of https://github.com/prusa3d/PrusaSlicer/issues/984 (https://github.com/prusa3d/PrusaSlicer/commit/6ea57817099ca72139bfa3a46fd1dd8a03749734).

bubnikv, 2021-12-08:
no response, closing

closed 2021-12-08 by bubnikv