prusa3d/PrusaSlicer #6998: Bug with support getting inside part

batata004, 2021-09-22:
Try to print this file with support: https://www.quemfazsite.com.br/temp/2.stl

You will see the green lines (support), get inside the part:

[image]

I painted the support as:

[image]

I have no idea why support is being print inside the part and this clearly is a bug.

rtyr, 2021-09-22:
Could you please try with 2.4.0-alpha1?

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1

Or please provide your 3MF project file, so we can test it with your exact settings.

batata004, 2021-09-22:
Of course my friend. Here is the file as you requested https://www.quemfazsite.com.br/temp/file.3mf

rtyr, 2021-09-22:
Thanks. It is OK in 2.4.0-alpha1, so we can close this issue.

[image]

closed 2021-09-22 by rtyr

batata004, 2021-09-22:
@rtyr  thanks my friend. I just installed the alpha version now but I cant find the "paint support" tool! Did you devs removed it? It was such a nice feature!