prusa3d/PrusaSlicer #6999: If raft layers > 0 then "Variable layer height not work".

Shama71, 2021-09-22:
### Version
2.4.0-alpha1

### Operating system type + version
Windows 11 64bit

### 3D printer brand / version + firmware version (if known)

### Behavior
If raft layers > 0 then "Variable layer height not work" and reset.

#### Project File (.3MF) where problem occurs

rtyr, 2021-09-22:
Reproduced, thank you. 

Works in 2.3.3.

46cv8, 2022-07-20:
Not sure if this is related but I get a hard crash if I enable a raft with 4 layers, then go to variable layer height and attempt to modify the first layer height. I'm using Linux and PrusaSlicer 2.4.2. The work-around for me is to disable the rift, modify with variable layer height, then re-enable the raft after and print. If you want any additional details let me know.

stepikovo, 2022-07-20:
@46cv8 It seems to be fixed in 2.5.0-alpha3, or at least I was not able to reproduce the error. Can you confirm?

46cv8, 2022-07-20:
I will try to confirm in the next day or so and get back to you. Thanks!