prusa3d/PrusaSlicer #7001: Missing text

reddawgesq, 2021-09-23:
2.3.0 Version

Windows 10 Operating system 

Prusa i3 MK3 3D printer 

Text that is present in the .STL/.OBJ files disappears when sliced. Other features just as small, such as the hash lines, slices correctly

[image]
uest?_

As sliced (and printed):

[image]

What gives? (BTW, associated /3MF file would not attach, messaded as "Not supported".

neophyl, 2021-09-23:
You have zip up the 3mf file same as you do on the forum reddawg.

rtyr, 2021-09-23:
The numbers are most likely too thin to be printed with set extrusion width. Please provide your 3MF project file.

reddawgesq, 2021-09-23:
Well, I am trying, but dragging the .3MF file into this box doesn't seem to work, which is odd in it's own right. Is there another way to attach it? Otherwise, the only other information I can provide is that this is Thingiverse.com item number4676171.

rtyr, 2021-09-23:
Found your model. The numbers are too thin, but you can solve that by enabling "Detect thin walls" parameter.

Can we close the issue?

[image]

reddawgesq, 2021-09-23:
Ok, I'll try that. Thanks.

reddawgesq, 2021-09-23:
Oops, that still doesn't work for me. Other ideas?

reddawgesq, 2021-09-23:
NM, got it sorted.

closed 2021-09-23 by reddawgesq

reddawgesq, 2021-09-26:
Made the change and printed it, but text numbers are still illegible. Thinking the .4mm nozzle is not going to handle it, will try again with a .2mm.

[dtw]<https://user-images.githubusercontent.com/36745189/134593766-c65ba303-d094-493b-8035-7b62a166f7b7.jpg>