prusa3d/PrusaSlicer #7003: max_layer_z is either not calculated/determined before processing placeholders, cause it to change at every layer during slicing, or it is pulling the layer_z value instead of the total height of the printed plate

cloud30000, 2021-09-24:
### Version
Version 2.4.0-alpha1+win64

### Operating system type + version
Windows 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
While trying to generate a Linear Advance Test Tower, both an offset needs to be used to begin the Linear Advance value as a negative and the same script needs to be usable with multiple different Layer Height profiles to get a value for each profile, so the total number of layers is unknown and changing.

Since the total print height of the model does not change, it made sense to use this static value and base a static offset on this value; however, I kept running into divide by zero errors, so I decided to simply record the "max_layer_z" placeholder at each layer change, at which point it became clear that the max_layer_z value was identical to the layer_z value.

This seems to indicate that it is either pulling the value of layer_z instead of the total height of the printed plate, or that it is being calculated and incremented while the placeholders are being evaluated and the gcode is being generated instead of calculating it prior to processing the placeholders like the total_layer_count.

BEFORE_LAYER_CHANGE Script which first caused this issue to be detected, being used on a Print which is exactly 171mm in height:
; Custom BEFORE_LAYER_CHANGE Code (MK3S+, {notes})
{if (layer_z>1)}M900 K{0.31*((layer_z)-(16))/max(1,((max_layer_z)-(16)))} ; After first layer, set Linear Advance to the Max Test Value multiplied by the percentage of the layers complete, with an offset to start at -0.03
{endif}; Layer number [layer_num], Layer height [layer_z]mm, Total Layers [total_layer_count], Max height [max_layer_z]mm
; Custom BEFORE_LAYER_CHANGE Code finished (MK3S+, {notes})

#### Project File (.3MF) where problem occurs
[Linear-Advance_Test_Tower_Bob_-0.03to0.31.zip](https://github.com/prusa3d/PrusaSlicer/files/7221431/Linear-Advance_Test_Tower_Bob_-0.03to0.31.zip)

bubnikv, 2021-09-24:
max_layer_z is set when starting to extrude a new object or support layer.
max_layer_z is not being updated from any of the custom G-code sections. That would require us to parse and analyze the custom G-code sections, which is non trivial and it may be time consuming.

cloud30000, 2021-10-09:
On further reading, it looks like the max_layer_z was not intended to provide the total print height, but just the current highest printed height to that point and will be identical to layer_z unless the nozzle was moved to a lower height for additional printing, such as when printing objects one at a time. As such, I will close this and add a featuer request for this placeholder instead.

closed 2021-10-09 by cloud30000