prusa3d/PrusaSlicer #7005: prusa-gcodeviewer changes modification time of the viewed gcode file

pfukan, 2021-09-24:
### Version
PrusaSlicer G-code Viewer
Version:   2.4.0-alpha2
Build:     PrusaSlicer G-code Viewer-2.4.0-alpha2-UNKNOWN

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 17,135MB
OpenGL installation
GL version:   4.6.0 NVIDIA 471.96
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### Behavior
When I open any *.gcode file in the prusa-gcodeviewer the time of this file is changed to the current system time.
The content of the file is unmodified.
I think the viewed file should remain untouched without a change of its attributes.

rtyr, 2021-09-27:
Related issue https://github.com/prusa3d/PrusaSlicer/issues/5079.

bubnikv, 2021-09-27:
Fixed with 07e7e11

This is a regression wrt. PrusaSlicer 2.4.0-alpha1 due to the G-code
processing optimization and parallelization.
Related to GCode Viewer changes files modified date in windows #5079

closed 2021-09-27 by bubnikv