prusa3d/PrusaSlicer #7007: Bad allocation error with large files

gusdleon, 2021-09-25:
### Version
2.3.2+win32

### Operating system type + version
Win 10 Pro 
Version: 21H1
Comp Ver: 19043.1237

### 3D printer brand / version + firmware version (if known)
N/A
but Original Prusa MK3s MMU2s

### Behavior
Opening large .stl files causes a bad allocation error in te program and there is 2 ways co cause the issue

1.-
* Open directly from file explorer the attached file PS will open but it will throw a bad allocation error and the file will not load

2.-
* Open PS and then import the .stl to the empty project [Ctrl+I] 
* the file will open but when you try to slice it  the program will immediately hard crash or show a bad allocation error then close

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
[This is the file with the problem hosted in my Google Drive](https://drive.google.com/file/d/11Cv42vflO9ZxbAfbOZDevWx7K8wIpwOw/view?usp=sharing)

#### NOTE:
I suppose the problem is I'm using the 32bit version of PS and can´t allocate enough memory to process the file but Idk

bubnikv, 2021-09-25:
Why don't you use the 64bit build of PrusaSlicer?

gusdleon, 2021-09-25:
I know I know... 
To my defense... I installed this version at launch day with the [drivers bundle](https://www.prusa3d.com/drivers/), and to my knowledge I had not option of 32bit and 64bit installer, the download page only give me a installer and it was the 32bit version of PS, and being honest, I never noticed I was using the 32 bit version. Sorry. Really Sorry. 😥

If it's really simply that, you can close this issue. thanks and sorry for the inconvenience 😓

lukasmatena, 2021-09-26:
> I installed this version at launch day

We should be the ones to apologize, actually. We released a buggy Windows installer with 2.3.2. It was fixed in matter of hours but you were apparently quicker.
Closing the issue.

closed 2021-09-26 by lukasmatena