prusa3d/PrusaSlicer #7009: Sending gcode to OctoPrint fails

GaryBoone, 2021-09-25:
### Version
Version 2.4.0-alpha2+x64
(Also occurred in 2.4 alpha1)

### Operating system type + version
macOS 11.6

### 3D printer brand / version + firmware version (if known)
Prusa mk3s+, firmware 3.10.0-4481

### Behavior
With PrusaSlicer 2.3.0, I have remote printing to OctoPrint working successfully. Using the same settings in the 2.4alpha, the "Send gcode to printer host" button fails, showing a popup that reads, "PrusaSlicer has encountered an error: Error uploading to print host: HTTP 400: ".

I can confirm that the printer host settings are the same: 
Host Type: OctoPrint, 
Hostname: [same as 2.3 setting]
API Key: [same as 2.3 setting]

The "Test" button results in "Connection to OctoPrint works correctly"

GaryBoone, 2021-09-25:
OctoPrint version info:

OctoPrint 1.6.1 
Python 2.7.16 
OctoPi 0.17.0

As mentioned above, this setup/API key works with PrusaSlicer 2.3.0.

xnk, 2021-09-26:
The exact same OctoPrint version works for me both on 2.4.0-alpha1 and 2 on macOS 10.15.7. Is this Big Sur specific? I simply copied the API key from the 2.3.3 install when installing alpha1 and kept on printing...

bubnikv, 2021-09-27:
"PrusaSlicer has encountered an error: Error uploading to print host: HTTP 400: ".

HTTP 400 is "Bad Request"

I looked up such an error on google:
https://github.com/OctoPrint/OctoPrint/issues/3674

Would you please inspect OctoPrint log files?