prusa3d/PrusaSlicer #7011: saving .3mf over an existing .3mf file increases the file size each time

ssill2, 2021-09-25:
### Version
_Version of PrusaSlicer used goes here_
build from master as of 09/25/2021
https://github.com/prusa3d/PrusaSlicer/commit/2356c303566b7b8ebcd8a8d8f66caca797eebe75

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
win 10 x64
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
i3 MK3S+

### Behavior
* _Describe the problem_
in latest master build doing a save fo the project or save as keeps dramatically increasing the size of the file.

In the attached .3mf file, simply saving it or saving it as a new file, say attempt_3.3mf makes the file go from 15MB to 44MB. Saving that 44MB file again makes it go to ~140MB.  each time the save operation takes substantially longer.

* _Steps needed to reproduce the problem_
Keep doing "save as" of the file to a new file name or even the same one.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
I would expect with no change to the file, simply saving it as a new file, the file size should remain constant
* _Actual Results_
File size seems to double with each save.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
you can see the result of each save or save as nearly doubles the size.  This is with no changes to the project, model, settings, etc.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[ps24 Hannibal Lecter Mask_sps_attempt_2.zip](https://github.com/prusa3d/PrusaSlicer/files/7230304/ps24.Hannibal.Lecter.Mask_sps_attempt_2.zip)

ssill2, 2021-09-26:
I can confirm that this problem does not happen on 2.3.3

ssill2, 2021-09-26:
so I'm trying to reproduce this today and it's not acting the same.

Now when I open the attempt_2.3mf and save as _3.3mf, it does go up in size, from ~15MB to 45MB.  But then saving it as other files from then on the the file size remains constant.  I was definitely seeing a roughly double in size each time I saved it last night
[image]

Interestingly, if I remove the two negative volume cylinders and save the file as a new file, the file goes back down to ~15MB from ~45MB.

bubnikv, 2021-09-27:
That is an unfortunate bug due to the admesh eradication refactoring, see the release log.
We shall fix it and likely release another alpha ASAP, otherwise buggy 3MFs will be produced.

ssill2, 2021-09-27:
sounds good thanks!

ssill2, 2021-09-27:
so I just synced up and built and saved the attempt_2 file as attempt_3 and that as attempt_4.
[image]

The size appeared to stabilize.  Interestingly, after saving it to _3 and removing the negative volumes, the size went up slightly.  it also got PS in a state where it said the model wouldn't fit on the plate.
See screenshot and attached 3mfs of 2.3 and 2.4 versions.

[image]

the ps23 version weighs in at less than 2MB. 
[ps23 Hannibal Lecter Mask_sps_attempt_4.zip](https://github.com/prusa3d/PrusaSlicer/files/7236270/ps23.Hannibal.Lecter.Mask_sps_attempt_4.zip)
it won't let me attached the .zip of the ps24 version, too large I think.  Interestingly though, if I take the ps23 file and open in 24 and save as a new file, the file size is nearly identical to the ps23 saved one, even slightly smaller.

[image]

ssill2, 2021-09-27:
The ps24 attempt 3 file that ps24 says won't fit on the build plate is too large to attach here.  clearly there's something going on in the file that makes it that large and make the slicer thing is too large for the build plate.

bubnikv, 2021-09-30:
Fixed with 
80f2853 and b76f66d2d23f3e6e0c6fc6eaab875082a6d52a46

For 3MFs generated by PrusaSlicer version >= "2.4.0-alpha1" and version < "2.4.0-alpha3",
the triangle meshes are processed to remove unreferenced vertices on loading. This should shrink the incorrectly generated 3MFs back to their original size.

closed 2021-09-30 by bubnikv