prusa3d/PrusaSlicer #7014: Automatically generated support structures are missing walls, such that parts of the support structure and/or support interfaces are unsupported

gpread, 2021-09-26:
2.4.0 Alpha 2+Win64

Windows 20H2 (19042.1165)

CR-10V3 (Modified) Marlin 2.0.8.2

New supports while nice :-) (since they don't seem to bleed through walls leaving redundant 1st layer material on the other side/edge) have missing elements, such that parts of the support are unsupported !!

1. Load the project file (attached below)
2. Make sure Print Settings->Support Material->Generate Support Material & Auto Generated Supports are enabled
3. Goto Print Settings->Support Material->Style and ensure "Grid" is selected
4. Slice the model, then examine the Feature Type view
5. Use the vertical layer selector to navigate to Layer 16 (1.57mm) and look at the support interface and support structures running inside the model along the X-Axis.

[image]
6. You can see that on one side there is only support material on the 1st layer with nothing at all to support the support material interface.

[image]
7. You can see on the other side that there is a just a single vertical support structure wall and the support material interface is a pure overhang.

[image]
8. Use the vertical layer selector to navigate to Layer 422 (39.40 mm) and look at the support interface and support structures running in the x-axis direction towards the rear of the model (also rear of the platter).  Similar support structures/issues can be seen.

[image]
9. Use the vertical layer selector to navigate to Layer 435 (40.50 mm) where you will see unsupported support interface structures above the support structures

[image]
10. Similarly Layer 439 (40.80mm) where you can also see unsupported support interface structures above the support structures

I also tried this sliced with the option Print Settings->Support Material->Style set to "Snug"  but there are similar problems.

From previous experience and experimentation with PS, I think it is worth considering decoupling PS supports grid internal resolution from the option "Pattern Spacing" (perhaps make it a setting of its own?).  Because with smaller features and/or smaller models its problematic to try to get less dense supports that accurately follow the models structure.  Furthermore, with the current coupling I found the support structures can fundamentally change in an unpredictable way when "Pattern Spacing" is changed where you are happy with how the supports were but want them a little less dense.

gpread, 2021-09-26:
[PS2_4_0_A2_SupportMaterial_unsupported_Issue#7014.zip](https://github.com/prusa3d/PrusaSlicer/files/7231646/PS2_4_0_A2_SupportMaterial_unsupported_Issue.7014.zip)

rtyr, 2021-09-27:
Reproduced on master, thanks.

bubnikv, 2021-10-07:
Fixed with 5de143f04fdb586490b28a4ac60654a1bb964a37

    Follow-up to 96b88f5b29f14d23f022d627b62e7c64883dcef8
    Fixed crash with zero support base spacing.
    The bug fix was not correct and it disabled the new "zig-zag" sparse
    infill generator, leading to GH issue #7014
    
The issue was introduced after alpha1 into alpha2.

closed 2021-10-07 by bubnikv

rtyr, 2021-10-12:
It is fixed on master. The fix will be included in the next release. Alpha3 still have this issue.