prusa3d/PrusaSlicer #7015: AMF files from FreeCAD no longer imports in 2.4.0-alpha2

xnk, 2021-09-26:
### Version
PrusaSlicer-2.4.0-alpha2+x64-202109241323

### Operating system type + version
macOS 10.15.7

### 3D printer brand / version + firmware version (if known)
Creality CR-10v3 / Marlin 2.0.9.1

### Behavior
AMF files exported from FreeCAD (0.19.2 in this case) no longer imports, instead a "The object is too small" error dialog shows up, and after that the z value is a very large negative value. Dialog and screenshot of 2.4.0-alpha2 and 2.3.3 side-by side:

[image]

[image]

Simple AMF test file attached here, I tried multiple files that used to work in 2.3.3, none of them imports in 2.4.0-alpha2:
[test-from-freecad.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/7232099/test-from-freecad.amf.zip)

xnk, 2021-09-26:
I should add that this particular object originally loads centered around the 0,0,0 point in 2.3.3 with a warning about an object outside the print area, but the object shows up and can be moved onto the print area without issues.

lukasmatena, 2021-10-01:
Broken due to admesh eradication refactoring in PrusaSlicer 2.4.0-alpha2, fixed with 1196251d5133aeec4d20ae36bf4a8462bcb8baf4
Also import of multi-material AMFs was broken due to the same refactoring and now fixed with afbbb36fa4c67ebbf9589ecc7ebb4c4db16c9dbb.
Closing.

closed 2021-10-01 by lukasmatena

bubnikv, 2021-10-01:
Broken due to admesh eradication refactoring in PrusaSlicer 2.4.0-alpha2, fixed with 1196251d5133aeec4d20ae36bf4a8462bcb8baf4

Also import of multi-material AMFs was broken due to the same refactoring and now fixed with afbbb36fa4c67ebbf9589ecc7ebb4c4db16c9dbb.

Closing.