prusa3d/PrusaSlicer #7017: Confirmation Box

ramosglauco, 2021-09-27:
### Version
2.4.0-alpha2

### Behavior
Hi devs! I would like to understand why I added this confirmation box after slicing. I can't ignore this message and every time I need to close it. This became a useless and boring task. Thank you.

[image]

_Is this a new feature request?_
Yes

foreachthing, 2021-09-27:
SolidWorks has on most dialog boxes a checkmark "don't show/ask anymore". I'd like to see that in PS too.

dartrax, 2021-09-28:
At least it should fade out automatically.

kocikdav, 2021-09-29:
Thank you everybody for your input. This notification was supposed to stay because we removed the status line (used to be on the bottom of PrusaSlicer window).  But it looks like it was not the best solution so it will now disappear after two seconds.
Removal of the status line is one of the latest changes so timing and appearance of notifications that inherited its functions might slightly change in next alfa / beta versions.

closed 2021-09-29 by kocikdav