prusa3d/PrusaSlicer #7024: Show a warning when using a Mini and adding M601

coaxial, 2021-09-28:
It is a long standing issue of the Mini that the M601 (pause print) code is just ignored over serial. There is an open issue from Mars 2020 (prusa3d/BuddyFirmware #283). 

I, like many others before me, have wasted time and filament expecting the printer to pause at a particular layer but ended up printing the whole part without pausing. 

As it is a known issue, the slicer should warn a user adding the M601 code when the printer type is a Mini. The warning should tell that M601 is ignored on that printer and either use M600 (filament change, not great but at least it pauses) or just be aware that it will be ignored.

rtyr, 2021-09-28:
I can understand your disappointment, but this is firmware issue, so it should be fixed in the firmware. 

Closing.

closed 2021-09-28 by rtyr

neophyl, 2021-09-28:
The question I would be asking, is that if the Mini does not respond to the M601 for Pause, why is that command included as part of the mini profile ?
If you go into the mini profile and delete the M601 in the Pause custom gcode field then that option is not available to the user.

As such you dont need a specific warning, you just need to make the default min profile correct.  Something Prusa could do quite easily just by editing their online profiles.  Wouldnt require any code changes.

rtyr, 2021-09-28:
@neophyl M601 is not ignored. This problem is only about printing via Octoprint afaik.

coaxial, 2021-09-28:
Right, I'm not saying this should be fixed in Prusa slicer because it is indeed a firmware issue. 

This is about adding a warning to Mini users who manually insert a M601 so that the user is made aware it will be ignored when printing over serial.

coaxial, 2021-09-28:
- Prusa doesn't seem interested in fixing it in the firmware
- Several people have been burned by it and have wasted filament on it
- The warning is not shown to all users but only to users who are slicing for the Mini
- The warning can have a checkbox to not show again or something to that effect

I agree this should be fixed in the firmware in the first place, but it isn't even on their roadmap. The warning would make the software more user friendly so that the users can avoid that pitfall.

neophyl, 2021-09-28:
The OP hasnt mentioned anything about Octoprint.  I dont have a mini so I dont know if the mini does respond to M601, all I was saying is that if it doesnt then sinply removing the M601 from the pause custom gcode in the printer profile would mean that the user cant use a function that the OP is saying doesnt work.  

I'd have to agree the best fix would be to make the firmware support the function though.