prusa3d/PrusaSlicer #7028: place holder total_layer_count wrong

Zuckme, 2021-09-29:
### Version
Version 2.3.3+win64

### Operating system type + version
Microsoft Windows [Version 10.0.19041.1237]

### 3D printer brand / version + firmware version (if known)
Ender 5

### Behavior
* _Steps needed to reproduce the problem_
Custom Gcode before layer change
M117 Layer {layer_num+1}/[total_layer_count]

* _Expected Results_
total_layer_count = 144

[image]

* _Actual Results_
total_layer_count = 171

see gcode attached. Last layer

`;LAYER_CHANGE
;Z:14.5
;HEIGHT:0.1
M117 Layer 144/171
G1 E0.00121 F3000.000
G92 E0
G1 Z14.500 F9000.000
G1 X107.919 Y115.017`

_Is this a new feature request?
No

#### Project File (.3MF) where problem occur
[total_layer_count wrong.zip](https://github.com/prusa3d/PrusaSlicer/files/7248467/total_layer_count.wrong.zip)
s

fe60, 2021-10-06:
The issue seems to be specific to the attached model. I guess the `total_layer_count` is calculated based on the object height not considering unprintable thin structures on the top of the model.
Reducing the extrusion width of (external) perimeters to 0.2mm increases the amount of printable layers and the gcode snippet of the last layer change looks like this:

```
...
;LAYER_CHANGE
;Z:16
;HEIGHT:0.1
;BEFORE_LAYER_CHANGE
G92 E0.0
;16
M117 Layer 159/171
...
```

Zuckme, 2021-10-19:
Either dev is very busy or time for me to try the current slic3r and see what happen there...

closed 2026-09-29 by prusaslicer-bot[bot]