prusa3d/PrusaSlicer #7029: Request: strip ALL Gcode comments.

NGC869, 2021-09-29:
Hi,
I suggest to introduce an option to completely strip ALL comments from the exported Gcode.
Thank you.

foreachthing, 2021-09-29:
You can do that with an easy python post processing script. If you need help, let me know!

lukasmatena, 2021-09-29:
@NGC869 Greetings fellow astronomy lover. Can you also disclose the reasons why you need it and which comments are in your way? Many are added so the gcode makes more sense when loaded later. I'd agree with @foreachthing that if there is a specific use case where they need to be stripped, it should be done by post-processing.

bubnikv, 2021-09-29:
If we remove all the G-code comments, we will not be able to display the G-code in G-code viewer. Namely, extrusion width / height and extrusion types are not part of the G-code per se, they need to be provided by the comments.

NGC869, 2021-09-30:
Hi all and thank you so much for your replies!

There are two main reason behind this request: some 3D printer goes crazy with gcode comments, and sometimes you don't want to share some informations (ex. detailed print settings) with third parts.

I recognize that you can achieve the same result with a post-processing script, I just thought that it wouldn't be a very complex task avoiding PS to print the already created informations.

Clear skies! ;)

lukasmatena, 2024-02-19:
We have introduced G-code substitutions in PrusaSlicer 2.4.1, which make it quite easy to deal with using regular expressions. Given that this is a corner case and vast majority of our users is likely not interested in the feature, I suppose it is how this should be handled. Closing.

lordton, 2024-05-23:
I have faced the similar issue, as i'm using a modified nc-studio cnc machine for printing.
The problem is that G-code substitutions work only for part of the g-code. I could replace comments in the beginning or end of the generated file.

For example using regexp     E.*      to substitute all extruder moves works flawlessly
But regexp         ; .*     or      ;.*     only removes comments in the body of g-code, ignoring the comments in header or footer sections. 

Perhaps they are added after the substituions fire in the pipeline..