prusa3d/PrusaSlicer #7030: Bed Size is limited to 1200mm x1200mm x1200mm

funny0facer, 2021-09-29:
### Version
2.3.3+win64

### Operating system type + version
Windows 10 Pro 21H1 
19043.1237

### Behavior
When setting up a bigger print bed than 1200mm x 1200mm x 1200mm, PrusaSlicer does not accept these values. Also a value in Origin bigger than 600mm is considered invalid.
Slic3r accepts values bigger than that, but I want to use the much nicer features of PrusaSlicer :)

Video of Behavior:
https://user-images.githubusercontent.com/75338570/135295074-3b3c0b13-0c24-4eb9-9429-9d166d59533d.mp4

### Printer
I am using a very big printer ca. 1300mm x 2500mm x 3000mm. See pictures:

[image]
[image]

bubnikv, 2021-09-29:
> Slic3r accepts values bigger than that, but I want to use the much nicer
features of PrusaSlicer

Slic3r accepts such values, but it may fail to perform gap fill, as the gap
fill algorithm works with 32bit coordinates.

We recommend to parametrize everything regarding your big printer in
centimeters.

funny0facer, 2021-09-29:
thank you for this explanation and workaround recommendation!

closed 2021-09-29 by funny0facer