prusa3d/PrusaSlicer #7032: No brim in supports PrusaSlicer-2.4.0-alpha2+win64

rschultzbr, 2021-09-29:
### Version
PrusaSlicer-2.4.0-alpha2+win64

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Ender 3 with 5 extruders on one hotend

### Behavior
PrusaSlicer-2.4.0-alpha2 is not generating brim on supports as 2.3.3 was
2.3.3 after slice
[image]

2.4.0 alpha2 after slice
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Hk.zip](https://github.com/prusa3d/PrusaSlicer/files/7254042/Hk.zip)

rtyr, 2021-09-29:
Intended behavior, see the changelog for 2.4.0-alpha1. Brim is newly calculated around the object, not around the 1st layer of the support and supports are printed over the brim on other layers.

It should be possible to use "First layer expansion" setting instead, but this setting is currently ignored (bug).

bubnikv, 2021-09-30:
The first layer expansion issue was fixed with 96f4d71c71cbb4055e1c67dcd12a52e848fac12b, it will be part of PrusaSlicer 2.4.0-alpha3.

closed 2021-09-30 by bubnikv