prusa3d/PrusaSlicer #7033: Tool missing (or not found) on prusaslicer 2.3.3

reos30, 2021-09-29:
### Version
_Version of PrusaSlicer 2.3.3

### Operating system type + version
Windows 7 integral edition

### 3D printer brand / version + firmware version (if known)
Créality CR10S pro V2

### Behavior
I don't have the orange tool "+" on the print's preview (like you can see on the picture attached), so I can't insert gcode for each layer changement.
I don't know if it is a bug or if I need to activate it somewhere (if yes, where please).

Thank you for your help.

[image]

reos30, 2021-09-29:
I found the trouble right after writing this post (it's been 2 day since I'm on it), it was because I cross the option "print individual object".
With this option cross, we don't have the "+" option on the print's preview.
With "print individual object" uncrossed, the "+" option appeared (bug or not, I don't know).

Thank you for reading.

Regard

rtyr, 2021-09-29:
Color change is not supported for sequential print. In 2.4.0-alpha, we newly indicate it by showing the "+" icon at the vertical slider of the g-code viewer grayed out and we also newly show a tooltip explaining why the action is not available.

[image]

Related issue https://github.com/prusa3d/PrusaSlicer/issues/5437.

Closing.

closed 2021-09-29 by rtyr