prusa3d/PrusaSlicer #7035: FR: Z Seam Options

kr15uk, 2021-09-30:
This is general Feature Request

Z Seams:

1. Paint on could have a "ruler tool", "drop on straight line tool" or point A to B automatic line when doing long straight seams.
2. Also click and select edge / two face intersection or so to create hidden seams.
3. Define XY location for a Z seam, at least XY max corner or something, not just a back.
4. In general "inteligent" auto seam like you get on Cura. I know "aligned" suppose to be that but it's really bad at it, like really really bad.
5. Straight seam to be straight and not wobbly.
[image]
6. More distinct Z seam location (like strong pink lines) in Preview mode to avoid disappointment hours later

Ta

Kris

ramosglauco, 2021-09-30:
The truth is that the seams need a lot of rework. I always have problems generating them. If I choose the "aligned" option, the seam is segmented (as in your image) if I choose the "rear" option, I still find seams on the front of the model. The seam paint tool is pretty bad too. It can be excruciating trying to make a straight line with the mouse. Everything you've requested has already been requested, and your topic will probably be closed. I would just like a prediction on any improvement in this part. For me, who produce decorative objects, a well-positioned seam makes a lot of difference.

ramosglauco, 2021-09-30:
But the main problem is that often when using seam paint only the start/end of the outer perimeter is moved. That is, the nozzle will need to travel a distance from the end of the inner perimeter to the beginning of the outer perimeter, using a retraction. In this way the retraction return happens just before the beginning of the outer perimeter, which is terrible from an aesthetic point of view, especially on large models and printers with the retraction not so well configured. The correct strategy would be that the starts/ends of all perimeters are always aligned, but we know that this thing doesn't happen. They could put the rework of the seams as a priority.

dartrax, 2021-10-01:
> But the main problem is that often when using seam paint only the start/end of the outer perimeter is moved. That is, the nozzle will need to travel a distance from the end of the inner perimeter to the beginning of the outer perimeter

Exactly. That is a problem.

A workaround getting more or less straight lines with paint on seam is to draw it while looking mainly from the top with only a small angle. You'll only have to move the mouse a very short way in a straight direction. Like this:
[image]
[image]
[image]

Another way is to enable the keyboard mouse. This way, you can use the number pad keys to steer the mouse. Here's how to enable that in windows: https://www.computerhope.com/issues/ch000542.htm

**However, my feature request would be:**
Use the shift key to lock the mouse in a way that it can only be moved up/down or left/right. Like it works when you move objects in MS Office.
I know that currently, Shift is used to delete what was painted before. I suggest using CTRL for that purpose. That is intuitive, because CTRL is already used in list boxes to delete the selection of the clicked item.

lukasmatena, 2024-02-19:
Related to #5970.

closed 2026-09-29 by prusaslicer-bot[bot]