prusa3d/PrusaSlicer #7036: Gap between inner and outer shells

svpcom, 2021-09-30:
### Version
PrusaSlicer-2.4.0-alpha1+linux-x64-202109011425

### Operating system type + version
Ubuntu 18.04.6

### 3D printer brand / version + firmware version (if known)
Custom printer based on Ultimaker2+ with Marlin 2.0.9.1 firmware

### Behavior
There is a gap between shells (see image attached). When slicing with Cura-4.11.0 no such gap found

#### Project File (.3MF) where problem occurs
[door_roll.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7260195/door_roll.3mf.zip)

Slicing in PrusaSlicer (gap exists):
[image]

Slicing in Cura (no gap):
[image]

rtyr, 2021-09-30:
You have disabled gap fill in your settings.

**Set non-zero speed here:**
[image]

**And enable fill gaps here (new parameter added in 2.4.0-alpha1):**
[image]

svpcom, 2021-10-01:
It help, thanks! But i'm confused with gap fill speed behavior - by default (for all other speeds) zero settings tuns on autospeed, but for gap it disables gap fill. It seems need to change behavior to be more consistent - zero gap fill will use autospeed and gap fill on/off should be only controlled by "fill gaps" checkbox.

dartrax, 2021-10-31:
> Changing the behavior may break somebody's old 3MF.

Can't we workaround this?
If gap_fill_enabled exists in 3MF config, gap_fill_speed = 0 means autospeed else it means gap fill disabled.

bubnikv, 2021-11-03:
I am not sure whether you want to enable autospeed for gap fill, because gap fill produces wildly varying extrusion rates. I believe autospeed is disabled for gap fill on purpose.

dartrax, 2021-11-03:
Ideally, the gap fill speed should be calculated in a way that extrusion rate remains constant...
Let's look at it the other way around: How would you decide for a single value for gap fill speed? The value must be low enough so that the widest extrusion path can be handled by the printer, but that would be slower than neccessary for smaller gap fills, or you have defined a max volumetric speed that you can rely on.
May be others have ideas here as I do not have many experiences with auto speed. I just agree to @svpcom, that the meaning of zero is error-prone at this place.

closed 2026-09-29 by prusaslicer-bot[bot]