prusa3d/PrusaSlicer #7038: 2.4.0-alpha 2 object rotation has display errors for y-axis rotator and does not display degree tooltip

evenprimes, 2021-09-30:
### Version
PrusaSlicer
Version:   2.4.0-alpha2+win64
Build:     PrusaSlicer-2.4.0-alpha2+win64-202109241329

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 8,354MB
OpenGL installation
GL version:   4.6.0 - Build 27.20.100.8935
Vendor:       Intel
Renderer:     Intel(R) UHD Graphics 620
GLSL version: 4.60 - Build 27.20.100.8935

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S, most recent release I believe

### Behavior
When rotating an object on the plater, the green (Y-Axis) rotation handle is always in the middle of the plater, no matter where the object being rotated might be.

When rotating, the degree tooltip no longer displays. The "snap" to 5 degree function still seems to work.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
[Front_Bed_Mount_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7262577/Front_Bed_Mount_3mf.zip)
[image]
[image]

bubnikv, 2021-10-01:
We were not able to reproduce your issue. Would you please help us to find a sequence of steps to reproduce the issue? Do you get the issue with the just released alpha3?
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha3

evenprimes, 2021-10-01:
I just tested with alpha3 and the issue seems to be resolved. Closing issue.

closed 2021-10-01 by evenprimes