prusa3d/PrusaSlicer #7039: 2.4.0-alpha2 Problems slicing MK3S to MK3S+ files

Saij, 2021-10-01:
### Version
2.4.0-alpha2+win64

### Operating system type + version
Windows 10 64 bit

### Behavior
I have downloaded the 3MF files for the MK3S to MK3S+ upgrade an wanted to print the provided E-axis 3MF file but the extruder cover does not slice correctly.

[image]

In the previous release version 2.3.1 everything slices correctly.

#### Project File (.3MF) where problem occurs
[E-axis_MK3StoMK3S+.zip](https://github.com/prusa3d/PrusaSlicer/files/7267512/E-axis_MK3StoMK3S%2B.zip)

bubnikv, 2021-10-01:
Fixed in PrusaSlicer 2.4.0-alpha3.

closed 2021-10-01 by bubnikv

Saij, 2021-10-01:
Thx - will wait for Alpha3 release